librespacefoundation/upsat-org #68: Ground loops (OBC - EPS)

zisi, 2016-06-07:


squarewavesbelow, 2016-06-07:
@zisi @manostheodoridis  please link to ti and maxim ground loop elimination strategy application notes.

ykoveos, 2016-06-08:
You shouldn't have any issues at this lengths/frequencies! -> Check connectors, cabling

In general:
1. Use small ferrite around power cables (common mode)
2. Twist signal cables with  ground, also use ferrite beads for EMI
3. Connect signal ground only at one side  (This depends on PCB layout, may become worst!)

In particular!
1. Set GPIO Pin speed to low to reduce ringing, EMI, etc.!

squarewavesbelow, 2016-06-08:
The problem was that we noticed that we cannot program the eps subsystem via st link with long cables, we can program it fine when we use shorter ones with the same setup (in need for active stlink /play trombone ). Agis also spotted st link blinking(as if it was in debug mode) while it was connected only from one side to eps swd. Also when measuring ground to ground resistance we did get great variations, @Johnkappa  maybe we should report them here for reference.

So apart from good practices maybe there exists a design flaw. EPS shares ground to all subsystems via the power bus and then OBC shares the same ground again to all subsystems via the intercommunication cables. @ykoveos  Do you think this might be problematic? would a star ground scheme be a safer way to go or it would probably be more immune to noise, given the use of no shielding in interconnect cables?

manostheodoridis, 2016-06-09:
Take a look https://www.maximintegrated.com/en/app-notes/index.mvp/id/5450

closed 2016-08-08 by ppapadeas